Stanford-BIS/pystorm #121: Cannot calibrate the default value for dac_syn_lk

arvoelke, 2018-12-13:
According to `pystorm.hal.DAC_DEFAULTS['DAC_SYN_LK']` the default value for the synapse leak is `10`. However, when I run:

```
MPLBACKEND=Agg python3 syn_tau.py 10 --num_trials=25
```

it gets to the very end of the calibration and then raises the error:

```
Traceback (most recent call last):
  File "syn_tau.py", line 528, in <module>
    HAL.add_calibration('synapse', synapse_cal, taus)
  File "/home/arvoelke/CTN/pystorm/pystorm/hal/hal.py", line 759, in add_calibration
    self.cdb.add_calibration(self.chip_name, cal_obj, cal_type, cal_data)
  File "/home/arvoelke/CTN/pystorm/pystorm/hal/cal_db.py", line 224, in add_calibration
    self.check_cal_pars(chip, cal_obj, cal_type)
  File "/home/arvoelke/CTN/pystorm/pystorm/hal/cal_db.py", line 202, in check_cal_pars
    raise ValueError(errstr)
ValueError: check_cal_pars(): you supplied calibration type tau_dac_10 which isn't known to CalibrationDB. All calibrations must be added to self.CAL_TYPES[synapse] supported calibration types for this object (synapse): ['tau_dac_1', 'tau_dac_2', 'tau_dac_4', 'tau_dac_8', 'tau_dac_16', 'high_bias_magnitude', 'pulse_width_dac_20', 'pulse_width_dac_21', 'pulse_width_dac_22', 'pulse_width_dac_23', 'pulse_width_dac_24', 'pulse_width_dac_25', 'pulse_width_dac_26', 'pulse_width_dac_27', 'pulse_width_dac_28', 'pulse_width_dac_29', 'pulse_width_dac_30', 'pulse_width_dac_31', 'pulse_width_dac_32', 'pulse_width_dac_33', 'pulse_width_dac_34', 'pulse_width_dac_35', 'pulse_width_dac_36', 'pulse_width_dac_37', 'pulse_width_dac_38', 'pulse_width_dac_39', 'pulse_width_dac_40']
```

Is there a way to get the calibration data for the default setting?

samfok, 2018-12-14:
This issue arises from our encoding of calibrations in `cal_db.py`, https://github.com/Stanford-BIS/pystorm/commit/95f4b4e1ceb91f220f04af2335e496f3da93bc10 adds the full range of possible `tau_dac` values

arvoelke, 2018-12-14:
Thank you! +1

closed 2018-12-14 by arvoelke